slipx06/sunsynk-power-flow-card #643: Add DC system load to card

JoshWobbles, 2025-02-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

There is no place to put 12v DC load information

### Expected behaviour

Separate AC and DC

### Possible Solutions

An optional additional entity branching from the inverter with power energy and current fields.

### Mode

Both

### Context / Reason

I also have a 12v DC system that I track the load of and would like to display it on the card coming off the inverter, currently I have it displayed as a nonessential load coming off the house but that isn't totally accurate.

SebastianTheCat, 2025-03-07:
Would it be possible to be able to choose if the load is W or %, my data gives % of load. It would also be nice if you could choose from where to where. Like if I have aux connected to Battery that affect everything. Like AUX is a input

sloniu82, 2025-04-22:
It would be great to have ability to vizualize DC loads taken directly from battery. 
In my case, I use Victron MPPT as a step-down converter - 48V from battery as PV input, and 12V on load/battey side.
This circut powers the 12V loads in my network cabinet - so it is totally offgrid - even if something wrong happen with the  inverter (of course, as long battery have enough power :) )