slipx06/sunsynk-power-flow-card #644: Full Card Showing Grey House 6.6.6

BlythMeister, 2025-02-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

Odd bug, in 6.6.6 a grey house with bold showing in load.

[image]

### Steps To Reproduce

Install 6.6.6 with full card

### Expected behaviour

Not to have odd icon

### Card Version

6.6.6

### Home Assistant Version

2025.2.5

### Configuration

```shell

```

### Relevant log output

```shell

```

martinisonline, 2025-03-01:
same issue here!

slipx06, 2025-03-01:
Thanks. Please share your config so I can track this this.

BlythMeister, 2025-03-01:
```
    type: custom:sunsynk-power-flow-card
    cardstyle: full
    show_solar: true
    battery:
      shutdown_soc: 20
      show_daily: true
      show_remaining_energy: true
      hide_soc: true
      animate: false
      linear_gradient: false
      count: 1
      auto_scale: true
      dynamic_colour: true
      max_power: 5000
      remaining_energy_to_shutdown: true
      colour:
        - 97
        - 195
        - 70
      charge_colour:
        - 97
        - 195
        - 70
      shutdown_soc_offgrid: 20
      soc_end_of_charge: 100
      path_threshold: 10
      soc_decimal_places: 0
    solar:
      show_daily: true
      mppts: 2
      pv1_name: FRONT
      pv2_name: REAR
      dynamic_colour: true
      auto_scale: true
      pv2_max_power: 2610
      pv1_max_power: 3045
      max_power: 5000
      efficiency: 3
      display_mode: 2
      off_threshold: 10
      colour:
        - 216
        - 229
        - 36
      custom_label: DAILY SOLAR / FORECAST SOLAR
    load:
      show_daily: true
      essential_name: HOME
      show_aux: false
      auto_scale: true
      dynamic_icon: true
      dynamic_colour: true
      load1_icon: ""
      additional_loads: 0
      max_power: 5000
      label_daily_load: HOME
      colour:
        - 103
        - 176
        - 255
      off_colour:
        - 97
        - 97
        - 97
      max_colour:
        - 103
        - 176
        - 255
      off_threshold: 10
      path_threshold: 10
    grid:
      show_daily_buy: true
      show_daily_sell: true
      show_nonessential: false
      auto_scale: true
      energy_cost_decimals: 2
      nonessential_icon: ""
      import_icon: ""
      export_icon: ""
      disconnected_icon: ""
      grid_name: "  "
      load1_icon: ""
      show_absolute: true
      off_threshold: 15
      max_power: 5000
      no_grid_colour:
        - 97
        - 97
        - 97
      colour:
        - 223
        - 98
        - 171
      export_colour:
        - 216
        - 229
        - 36
      grid_off_colour:
        - 97
        - 97
        - 97
    entities:
      use_timer_248: sensor.inverter_use_timer
      priority_load_243: sensor.inverter_priority_load
      inverter_voltage_154: sensor.solarsynk_inverter_voltage
      load_frequency_192: sensor.solarsynk_load_frequency
      inverter_current_164: sensor.solarsynk_inverter_current
      inverter_power_175: sensor.solarsynk_inverter_power
      grid_connected_status_194: sensor.solarsynk_grid_connected_status
      inverter_status_59: sensor.solarsynk_overall_state
      day_battery_charge_70: sensor.solarsynk_day_battery_charge
      day_battery_discharge_71: sensor.solarsynk_day_battery_discharge
      battery_voltage_183: sensor.solarsynk_battery_voltage
      battery_soc_184: sensor.solarsynk_battery_soc
      battery_power_190: sensor.solarsynk_battery_power
      battery_current_191: sensor.solarsynk_battery_current
      grid_power_169: sensor.solarsynk_grid_power
      day_grid_import_76: sensor.solarsynk_day_grid_import
      day_grid_export_77: sensor.solarsynk_day_grid_export
      grid_ct_power_172: sensor.solarsynk_grid_power
      day_load_energy_84: sensor.solarsynk_day_load_energy
      day_pv_energy_108: sensor.solarsynk_day_pv_energy
      pv1_power_186: sensor.solarsynk_pv1_power
      pv1_voltage_109: sensor.solarsynk_pv1_voltage
      pv1_current_110: sensor.solarsynk_pv1_current
      pv2_voltage_111: sensor.solarsynk_pv2_voltage
      pv2_current_112: sensor.solarsynk_pv2_current
      battery_rated_capacity: sensor.solarsynk_battery_capacity
      battery_temp_182: sensor.solarsynk_battery_temperature
      prog1_time: sensor.solarsynk_prog1_time
      prog1_capacity: sensor.solarsynk_prog1_capacity
      prog1_charge: sensor.solarsynk_prog1_charge
      prog2_time: sensor.solarsynk_prog2_time
      prog2_capacity: sensor.solarsynk_prog2_capacity
      prog2_charge: sensor.solarsynk_prog2_charge
      prog3_time: sensor.solarsynk_prog3_time
      prog3_capacity: sensor.solarsynk_prog3_capacity
      prog3_charge: sensor.solarsynk_prog3_charge
      prog4_time: sensor.solarsynk_prog4_time
      prog4_capacity: sensor.solarsynk_prog4_capacity
      prog4_charge: sensor.solarsynk_prog4_charge
      prog5_time: sensor.solarsynk_prog5_time
      prog5_capacity: sensor.solarsynk_prog5_capacity
      prog5_charge: sensor.solarsynk_prog5_charge
      prog6_time: sensor.solarsynk_prog6_time
      prog6_capacity: sensor.solarsynk_prog6_capacity
      grid_voltage: sensor.solarsynk_grid_voltage
      dc_transformer_temp_90: sensor.solarsynk_dc_temperature
      radiator_temp_91: sensor.solarsynk_ac_temperature
      battery_status: sensor.solarsynk_battery1_status
      pv2_power_187: sensor.solarsynk_pv2_power
      prog6_charge: sensor.solarsynk_prog6_charge
      load_power_L1: sensor.solarsynk_load_power
      remaining_solar: sensor.solcast_pv_forecast_forecast_today
      environment_temp: sensor.garden_thermometer
      total_pv_generation: sensor.vmeter_total_pv
      grid_ct_power_total: sensor.solarsynk_grid_power
    large_font: true
    show_battery: true
    show_grid: true
    decimal_places_energy: 1
    dynamic_line_width: false
    inverter:
      autarky: "no"
      modern: true
      model: sunsynk
      auto_scale: true
      three_phase: false
      colour:
        - 248
        - 179
        - 40
    grid_options:
      columns: full
      rows: auto
    wide: false
    decimal_places: 1
    max_line_width: 10
    min_line_width: 1
```

slipx06, 2025-03-01:
Ok sorry about that. Pushing a fix now

slipx06, 2025-03-01:
Please test https://github.com/slipx06/sunsynk-power-flow-card/releases/tag/v6.6.7

November481, 2025-03-01:
[image]
Same issue, waiting for new update

BlythMeister, 2025-03-01:
Works for me.

Did you remember to clear your browser cache as the install prompt says?

slipx06, 2025-03-01:
Remember to clear your front end cache

closed 2025-03-01 by slipx06

November481, 2025-03-01:
Fixed, i use HA for updates, dont know how to update with your files, in HA i always get update. Thanks