slipx06/sunsynk-power-flow-card #651: day_pv_energy_108 not "clickable"

albundy0815, 2025-03-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

In the dashboard nearly all values are clickable, to get more details on it (entities are shown) but not for day_pv_energy_108. 
If I click on it, it's not related to the entity history.

### Expected behaviour

It would be nice, to be able to open the details of the entity.

### Mode

Both

### Context / Reason

to get a better overview, it would be great to have this feature.

albundy0815, 2025-03-18:
[image]

slipx06, 2025-03-18:
Hi. That entity is only clickable if you provide one in your config. i.e. pv_total:

https://slipx06.github.io/sunsynk-power-flow-card/configuration.html#solar-entities

If not the card used its own internal logic and there is no state history maintained.

closed 2025-03-19 by albundy0815