slipx06/sunsynk-power-flow-card #657: SigEnergy Image

Roving-Ronin, 2025-03-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

Hi @slipx06,

Would you please be able to add an option for having a SigEnergy Inverter+Battery system appear in the card?
i.e.: 

### Expected behaviour

inverter:
  modern: false
  model: sigenergy

### Possible Solutions

[image]

### Mode

Both

### Context / Reason

Add support for showing SigEnergy ESS upon the card.

slipx06, 2025-04-05:
@Roving-Ronin  I've added the image but it does not scale well. Would be better if we had a vector image.  If you find something better let me know

Roving-Ronin, 2025-04-06:
Hi @slipx06 

Are these any better ? https://www.sigenergy.com/en/support/files/371 

This includes all the different 'sizes', given SigEnergy is modular (1 inverter on the top, then up to 5 [if install the 25kW DC charger] or 6 battery modules then stack below the inverter).

slipx06, 2025-04-06:
Not really, but I've updated it in the new fix. It creates a very pixelated image when I reduce the size.

Roving-Ronin, 2025-09-04:
Hi @slipx06 following on from this initial support for Sigenergy, would you please be able to have a look at these PULL requests to:

- Add mappings for Sigenergy inverters/batteries: https://github.com/slipx06/sunsynk-power-flow-card/pull/720

- Add the documentation for Sigenergy: https://github.com/slipx06/sunsynk-power-flow-card/pull/723

Many Thanks !

commit ad2cc5b7 (2025-04-05): feat: added new inverter models for EASun, Ferroamp, Sigenergy and Linky closes #657, #579
feat: six essential loads can now be displayed on the card without the need for wide screen display mode if certain conditions are met i.e. mppt <=3 on lite card and !show_aux on full card etc closes #635