slipx06/sunsynk-power-flow-card #661: Card disappear after update

Skyden72, 2025-04-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

Hi Slipx,

After updating to 2025.4.1 the card disappeared. Still shows in Hacs, and I tried re-installing. I even tried deleting and reinstalling it, but it didn't work.

### Expected behaviour

Must show up

### Card Version

6.6.9

### Home Assistant Version

2025.4.1

### Configuration

```shell

```

### Relevant log output

```shell

```

slipx06, 2025-04-05:
Hi @Skyden72 . I cant reproduce this. What Dashboard are you using. Have you tried to clear your frontend cache? Also any console logs? Bring up developer tools F12 and look for any errors

[image]

Kepasub, 2025-04-05:
Hi Skyden72 and slipx06
I apologize if what happened to me while updating is unrelated to your comment.
When updating, the screen turned red and I was in "Recovery Mode Enabled."
In this mode, I can't even restore a previous version.
And there's a very long line of what appears to be an "Error Log," which I can't read.
I can't follow the steps slipx06 suggests because I only know how to program in graphical mode, and I'm afraid of messing things up.

I also get these two notifications:

Config validating
Config error. See the logs for details.

Home Assistant is running in recovery mode. Check the error log to see what went wrong.

But I don't know what to do with them either.

Kepasub, 2025-04-05:
I apologize again. It's now resolved.
In /homeassistant/config yaml, I created a line with the text homeassistant. I deleted that text and everything seems to be fixed. I apologize again.

Skyden72, 2025-04-05:
> Hi [@Skyden72](https://github.com/Skyden72) . I cant reproduce this. What Dashboard are you using. Have you tried to clear your frontend cache? Also any console logs? Bring up developer tools F12 and look for any errors
> 
> [image]

Thanks Slipx, I tried what you suggested with no joy. I reverted Hass to the pre-2025.4 backup, and that sorted it out. The new update broke a lot of my custom cards, dashboards, etc. Will wait for updates before I upgrade again. This is the first time this has happened. Been using Hass for two years now.

closed 2025-04-05 by slipx06