slipx06/sunsynk-power-flow-card #666: Essential Load names not centered

pvprodk, 2025-04-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

Love the card, and the frequent quality upgrades it recieves!

This is not any kind of dealbreaker, but just a minor cosmetic request.

Would it be possible to center all 6 load names, currently 3 of the 6 names are correctly centered:

[image]

### Expected behaviour

Centered load names

### Card Version

6.7.4

### Home Assistant Version

2025.4.1

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: lite
show_solar: true
battery:
  energy: 61440
  shutdown_soc: 10
  show_daily: true
  shutdown_soc_offgrid: false
  hide_soc: true
  show_absolute: false
  linear_gradient: false
  animate: false
  colour: mediumseagreen
  dynamic_colour: false
  count: 1
  invert_power: false
  auto_scale: false
  max_power: 20000
  soc_decimal_places: 0
solar:
  show_daily: true
  display_mode: 2
  mppts: 2
  pv1_max_power: 11300
  max_power: 16100
  pv1_name: Høj/Garage
  pv2_name: Flad
  pv2_max_power: 4800
load:
  show_daily: true
  show_aux: false
  max_power: 15000
  essential_name: Load-side
  additional_loads: 6
  load1_name: Easee
  load2_name: Pool
  load3_name: VM
  load4_name: Opvsk.
  load5_name: PC
  load6_name: TT
  dynamic_icon: true
  dynamic_colour: true
  colour: palevioletred
  off_threshold: 20
  off_colour: grey
  auto_scale: true
  invert_load: false
  load1_icon: ""
  load2_icon: ""
  load3_icon: ""
  load4_icon: ""
  load5_icon: ""
  load6_icon: ""
grid:
  show_daily_buy: true
  show_daily_sell: true
  show_nonessential: false
  max_power: 15000
  nonessential_name: Grid-side
panel_mode: true
large_font: true
show_battery: true
show_grid: true
inverter:
  three_phase: true
  model: deye
  modern: true
  autarky: "no"
  colour: tan
entities:
  use_timer_248: "no"
  priority_load_243: "no"
  inverter_voltage_154: sensor.deye_grid_voltage_l1
  inverter_voltage_L2: sensor.deye_grid_voltage_l2
  inverter_voltage_L3: sensor.deye_grid_voltage_l3
  load_frequency_192: sensor.deye_inverter_frequency
  inverter_current_164: sensor.deye_inverter_total_current_l1
  inverter_current_L2: sensor.deye_inverter_total_current_l2
  inverter_current_L3: sensor.deye_inverter_total_current_l3
  inverter_power_175: sensor.deye_total_power_output
  grid_connected_status_194: binary_sensor.deye_grid_relay_status
  inverter_status_59: sensor.deye_running_status
  day_battery_charge_70: sensor.deye_daily_battery_charge
  day_battery_discharge_71: sensor.deye_daily_battery_discharge
  battery_voltage_183: sensor.deye_battery_voltage
  battery_soc_184: sensor.deye_battery_capacity
  battery_current_191: sensor.deye_battery_output_current
  battery_temp_182: sensor.deye_battery_temperature
  grid_power_169: sensor.deye_total_power_grid_port
  energy_cost_buy: sensor.energi_data_service_kob
  energy_cost_sell: sensor.energi_data_service_salg
  day_grid_import_76: sensor.grid_daily_import
  day_grid_export_77: sensor.grid_daily_export
  grid_ct_power_172: sensor.grid_active_power_phase_1
  grid_ct_power_L2: sensor.grid_active_power_phase_2
  grid_ct_power_L3: sensor.grid_active_power_phase_3
  day_load_energy_84: sensor.daily_energy_consumption
  essential_power: sensor.deye_inverter_load_total_power
  essential_load1: sensor.easee_power
  essential_load2: sensor.udespa_power
  essential_load3: sensor.minipmg3_vaskemaskine_power
  essential_load4: sensor.opvaskemaskine_power
  aux_power_166: none
  remaining_solar: sensor.solcast_pv_forecast_forecast_remaining_today
  day_pv_energy_108: sensor.deye_daily_production
  pv1_power_186: sensor.deye_pv1_power
  pv1_voltage_109: sensor.deye_pv1_voltage
  pv1_current_110: sensor.deye_pv1_current
  pv2_power_187: sensor.deye_pv2_power
  pv2_voltage_111: sensor.deye_pv2_voltage
  pv2_current_112: sensor.deye_pv2_current
  pv_total: sensor.deye_pv_power
  solar_sell_247: switch.deye_solar_sell
  environment_temp: sensor.openweathermap_temperature
  radiator_temp_91: sensor.deye_heat_sink_temperature
  pv3_power_188: sensor.growatt_pv1_active_power
  battery_power_190: sensor.deye_battery_output_power
  battery2_power_190: sensor.seplos_battery_bank_1_power
  battery2_current_191: sensor.seplos_battery_bank_1_current
  battery2_temp_182: sensor.seplos_battery_bank_1_temperature_1
  battery2_voltage_183: sensor.seplos_battery_bank_1_total_voltage
  battery2_soc_184: sensor.seplos_battery_bank_1_state_of_charge
  battery2_soh: sensor.seplos_battery_bank_1_state_of_health
  essential_load6: sensor.minipmg3_torretumbler_power
  essential_load5: sensor.jonas_computer_power
wide: true
grid_options:
  columns: full
```

### Relevant log output

```shell

```

slipx06, 2025-04-12:
Hi. Thanks. They are not centered in order to handle longer names and avoid overlap i.e. left and right alignment.  It was introduced before I had the option to display 6 loads.  I'll see if I can improve it.

[image]

slipx06, 2025-04-12:
The alignment was inconsistent so I've fixed now. Everything to the right of the mid line is aligned to the left and everything to the left is aligned to the right

[image]

closed 2025-04-12 by slipx06

pvprodk, 2025-04-12:
Sorry to say, but I dont like it at all. I would like it to be centered on all loads, icons as well (although I dont use those at the moment)

This is not good
[image]

slipx06, 2025-04-12:
You can centre them manually by adding white space ( \_)after the name i.e.

`load5_name: "PC\_\_\_"`

commit 9dbba223 (2025-04-12): fix: alignment of essential load items closes #666