slipx06/sunsynk-power-flow-card #672: Retrive entities

MBRetrofit, 2025-05-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

Everything works smoothly,

### Expected behaviour

Is it any way to use entities from this Card? Like Time to charge/discharge?

### Mode

lite

### Context / Reason

I would like to use this into automation and badges

slipx06, 2025-05-04:
Hello. No its not possible, but you can create your own sensos to calculate those values.  If you scroll down on https://github.com/slipx06/Sunsynk-Home-Assistant-Dash there is some example code for remaining battery time

MBRetrofit, 2025-05-04:
Yeah, I already tried, modified, for several days...is deprecated for the latest HA template specification, like "state" instead "value_tenplate" and so on. Not working at all, for me....as I have almost the same entities for a Easun SMG II, Although the card is OK...

slipx06, 2025-05-04:
I'm not seeing any errors. Maybe share your code?

Did you add `sensor: !include sensors.yaml` to your HA `configuration.yaml` file. and paste the code below into `sensors.yaml`? You can use this as a base to modify for your sensors and logic i.e. remove the sellTime etc.

```
- platform: template
  sensors:
    battery_cap:
      friendly_name: "Battery Capacity"
      value_template: >
        {% set grid_online = states('binary_sensor.sunsynk_grid_connected_status') %}
        {% if grid_online  == 'off'%}
          {{ states('sensor.sunsynk_battery_capacity_shutdown') | int }}
        {% else %}
          {% set now = strptime(as_timestamp(now()) | timestamp_custom('%H:%M'), '%H:%M') %}
          {% set sellTime1 = strptime(states('sensor.sunsynk_time_slot_1'), '%H:%M') %}
          {% set sellTime2 = strptime(states('sensor.sunsynk_time_slot_2'), '%H:%M') %}
          {% set sellTime3 = strptime(states('sensor.sunsynk_time_slot_3'), '%H:%M') %}
          {% set sellTime4 = strptime(states('sensor.sunsynk_time_slot_4'), '%H:%M') %}
          {% set sellTime5 = strptime(states('sensor.sunsynk_time_slot_5'), '%H:%M') %}
          {% set sellTime6 = strptime(states('sensor.sunsynk_time_slot_6'), '%H:%M') %}
          {% if now >= sellTime1 and now < sellTime2 %}
            {{ states('number.sunsynk_prog1_capacity') | int }}
          {% elif now >= sellTime2 and now < sellTime3 %}
            {{ states('number.sunsynk_prog2_capacity') | int }}
          {% elif now >= sellTime3 and now < sellTime4 %}
            {{ states('number.sunsynk_prog3_capacity') | int }}
          {% elif now >= sellTime4 and now < sellTime5 %}
            {{ states('number.sunsynk_prog4_capacity') | int }}
          {% elif now >= sellTime5 and now < sellTime6 %}
            {{ states('number.sunsynk_prog5_capacity') | int }}
          {% elif now >= sellTime6 or now < sellTime1 %}
            {{ states('number.sunsynk_prog6_capacity') | int }}
          {% else %}
            {{ states('sensor.sunsynk_battery_capacity_shutdown') | int }}
          {% endif %}
        {% endif %}
    soc_battery_time_left:
      friendly_name: "Battery Depletion Seconds"
      unit_of_measurement: Seconds
      value_template: >
        {% set state = states('sensor.sunsynk_battery_power') | int %}
        {% set cap = states('sensor.battery_cap') | float %}
        {% if state == 0 -%}
        {{ ((((states('sensor.sunsynk_battery_soc') | float - cap) /100) * 15960) / (1) * 60 * 60 ) | timestamp_custom('%s', 0) }}
        {%- else -%}
        {{ ((((states('sensor.sunsynk_battery_soc') | float - cap) /100) * 15960) / (states('sensor.sunsynk_battery_power') | float) * 60 * 60 ) | timestamp_custom('%s', 0) }}
        {%- endif %}
    soc_battery_time_left_friendly:
      friendly_name: "Battery Depletion Time"
      value_template: >
        {% set state = states('sensor.sunsynk_battery_power') | int %}
        {% if state > 0 -%}
        {%- set time = states('sensor.soc_battery_time_left') | int %}
        {%- set minutes = ((time % 3600) // 60) %}
        {%- set minutes = '{} minutes'.format(minutes) if minutes > 0 else '' %}
        {%- set hours = ((time % 86400) // 3600) %}
        {%- set hours = '{} hours, '.format(hours) if hours > 0 else '' %}
        {%- set days = (time // 86400) %}
        {%- set days = '{} day, '.format(days) if days > 0 else '' %}
        {{ 'Less than 1 minute' if time < 60 else days + hours + minutes }}
        {%- else -%}
        {{ 'Charging' }}
        {%- endif %}
    battery_charging_time_left:
      friendly_name: "Battery Charging Time Left"
      unit_of_measurement: Seconds
      value_template: >
        {% set power = states('sensor.sunsynk_battery_power') | float %}
        {% set soc = states('sensor.sunsynk_battery_soc') | float %}
        {% set cap = states('sensor.battery_cap') | float %}
        {% if power < 0 %}
          {% if soc < cap %}
            {{ ((((cap - soc) / 100) * 15960) / (-power) * 60 * 60) | int }}
          {% else %}
            {{ ((((100 - soc) / 100) * 15960) / (-power) * 60 * 60) | int }}
          {% endif %}
        {% else %}
          0
        {% endif %}
    battery_charging_time_left_friendly:
      friendly_name: "Battery Charging Time"
      value_template: >
        {% set state = states('sensor.sunsynk_battery_power') | int %}
        {% if state < 0 -%}
          {%- set time = states('sensor.battery_charging_time_left') | int %}
          {%- set minutes = ((time % 3600) // 60) %}
          {%- set minutes = '{} min'.format(minutes) if minutes > 0 else '' %}
          {%- set hours = ((time % 86400) // 3600) %}
          {%- set hours = '{} hrs, '.format(hours) if hours > 0 else '' %}
          {%- set days = (time // 86400) %}
          {%- set days = '{} day, '.format(days) if days > 0 else '' %}
          {{ 'Floating' if time < 60 else days + hours + minutes }}
        {%- else -%}
          {{ 'Discharging' }}
        {%- endif %}
    markdown_battery_charge_time_left:
      friendly_name: "Markdown Battery Charging Time"
      value_template: >
        {% if states('sensor.sunsynk_battery_soc') | float < states('sensor.battery_cap') | float %}
          {{ states('sensor.battery_cap') | float | round(0) }}
        {% else %}
          100
          {% endif %}
    markdown_discharge_time:
      friendly_name: "Markdown Discharge Time"
      value_template: >
        {% set now = as_timestamp(now()) %}
        {% set add = states('sensor.soc_battery_time_left') | int %}
        {% set future_time = now + add %}
          {{ future_time | timestamp_custom('%H:%M') }}
    markdown_charge_time:
      friendly_name: "Markdown Charging Time"
      value_template: >
        {% set now = as_timestamp(now()) %}
        {% set add = states('sensor.battery_charging_time_left') | int %}
        {% set future_time = now + add %}
          {{ future_time | timestamp_custom('%H:%M') }}
```