slipx06/sunsynk-power-flow-card #701: efficiency not displayed

Loic691, 2025-06-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

Hi all
Thanks for this great integration.
I have a Deye micro inverter with 2 PV each on 1 MPPT.
The PV power, voltage and current are clearly displayed.
I have configured PV Max to 1000, PV1 max to 500 et PV2 max to 500.
But there is no diaplayed value. See capture

I have help to know too what is the ratio value at 100% ? Is this the difference between total PV power and Inverter power (Deye have the loss power).

Thanks 

[image]

### Expected behaviour

Efficency displayed

### Card Version

v6.8.0

### Home Assistant Version

2025.6.1

### Configuration

```shell

```

### Relevant log output

```shell

```

slipx06, 2025-06-19:
Hi

Have you set your efficiency mode correctly? Its under solar options.

[image]

or you can switch to the code editor and change

```
solar:
  efficiency: 3
```

slipx06, 2025-06-19:
All the options are described here

https://slipx06.github.io/sunsynk-power-flow-card/configuration.html#solar

closed 2025-06-20 by slipx06