slipx06/sunsynk-power-flow-card #722: Battery Options fehlen

tapian1, 2025-08-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

Seit dem letzten Update fehlen mir die "Battery Options" Ich kann diese nur noch im CodeEditor bearbeiten. Allerdings kenne ich mich mit codes nicht gut aus, daher wäre es super wenn die Batterieoptionen wieder als DropDown menü vorhanden wären :)

### Expected behaviour

Battery Options wieder sichtbar werden lassen

### Card Version

7.0.0

### Home Assistant Version

2025.08.1

### Configuration

```shell
type: custom:sunsynk-power-flow-card
cardstyle: compact
show_solar: true
battery:
  show_daily: false
  show_absolute: true
  auto_scale: true
  hide_soc: true
  dynamic_colour: false
  linear_gradient: false
  animate: false
  show_remaining_energy: false
  remaining_energy_to_shutdown: false
  invert_flow: true
  invert_power: false
  colour: "#ff5100"
  charge_colour: "#00ff00"
  count: 1
  path_threshold: 0
solar:
  show_daily: true
  mppts: 3
  dynamic_colour: true
  pv1_max_power: 3680
  display_mode: 1
  pv2_max_power: 1
  efficiency: 3
  custom_label: Hausanlage
  colour: "#ffa500"
  auto_scale: false
  max_power: 8280
  pv1_name: Hochbeet (3,4)
  invert_flow: false
  pv3_name: Parkplatz (1,2)
  pv2_name: Garten (5)
  pv3_max_power: 4600
load:
  show_daily: false
  label_daily_load: Verbraucht
  colour: "#ffbb00"
  additional_loads: 0
  load1_name: Gewächshaus
  load2_name: Haus
  load2_icon: mdi:home
  load1_icon: mdi:home-silo
  dynamic_icon: true
  dynamic_colour: true
  auto_scale: false
  invert_load: false
  load3_name: Heizstab
  load3_icon: mdi:heat-wave
  show_aux: false
  invert_flow: false
grid:
  show_daily_buy: true
  show_daily_sell: false
  show_nonessential: true
  label_daily_grid_sell: " "
  label_daily_grid_buy: " Netzbezug"
  grid_name: " "
entities:
  inverter_power_175: sensor.sunsynk_inverter_power
  day_battery_charge_70: sensor.daily_battery_charge
  day_battery_discharge_71: sensor.daily_battery_discharge
  battery_soc_184: sensor.battery_level
  grid_power_169: sensor.sunsynk_grid_power
  day_grid_import_76: sensor.daily_imported_energy
  day_grid_export_77: sensor.daily_exported_energy
  day_pv_energy_108: sensor.daily_pv_generation
  pv_total: sensor.total_dc_power
  grid_ct_power_172: sensor.meter_active_power
  battery_power_190: sensor.laden_oder_entladen_sg
  battery2_soc_184: sensor.wr_3_battery_charge_2
  essential_load1: sensor.delta_echtzeitverbrauch
  essential_load2: sensor.load_power
  total_pv_generation: sensor.cee_tagesverbrauch
  pv1_power_186: sensor.mppt2_power
  essential_load3: number.vebrauch_elwa_power_elwa_2
  pv3_power_188: sensor.mppt1_power
  essential_power: sensor.load_power
large_font: false
wide: true
inverter:
  autarky: "no"
  model: linky
  modern: true
battery2:
  show_absolute: true
  auto_scale: true
  invert_power: true
  linear_gradient: true
  colour: "#004400"
  hide_soc: true
  animate: false
  show_remaining_energy: false
  remaining_energy_to_shutdown: false
  dynamic_colour: false
  invert_flow: false
title_colour: "#00ff1e"
title_size: "5"
dynamic_line_width: true
max_line_width: 3
min_line_width: 1
grid_options:
  columns: 24
  rows: 6
card_mod:
  style: |
    ha-card {
      border: 3px solid blac;
      border-radius: 10px;
    }
```

### Relevant log output

```shell

```

slipx06, 2025-08-26:
Hi Do you not see this? Have you reset you browser frontend cache?

[image]

tapian1, 2025-08-27:
> OK I think I found the issue, Please toggle this off and then on and see
> if that fixes the problem
> image.png (view on web)

Vielen dank, dass war der Fehler. Entschuldige bitte.